end2endzone/win32Arduino #15: win32Arduino on Linux

sidey79, 2018-03-28:
Is there any chance to get rid of the win32 specific libs / functions used in this lib?

I wanted to run some integration tests with travic ci but there are serval things missing.

end2endzone, 2018-04-12:
That's a good idea and I had this planned for a long time. I am locally working on porting all scripts from /ci/appveyor to /ci/ant to have cross-Platform scripts. However, I do not have much experience programming in Linux (I do own a Raspberry Pi & a Ubuntu partition).

I do not know how many sections of the library would be impacted by the change but I think it would be possible without too much work. Nothing that a google search can't fix.

end2endzone, 2018-04-12:
I think it would be a matter of wrapping each function implementation with `#ifdef` statements. See [stackoverflow](https://stackoverflow.com/a/18729350) for details.

end2endzone, 2018-04-12:
I would be glad if you give me a hand on this and help me porting the code to linux. Feel free to send me a pull request.

sidey79, 2018-04-12:
I am not so familiar with all the win32 libs.

I think, there are std cpp libs which are available crossplattform.

But win32arduino usues

end2endzone, 2018-04-15:
@sidey79 I would appreciate your help for issue #17.

end2endzone, 2018-04-15:
@sidey79 The code is now able to compile without errors on the Raspberry Pi. However, I am not able to compile on Travis CI.

The following error is displayed in the [Travis CI build log](https://travis-ci.org/end2endzone/win32Arduino):
```
     [exec] /home/travis/build/end2endzone/win32Arduino/src/win32Arduino/arduino.cpp: In function ‘void randomSeed(int16_t)’:
     [exec] /home/travis/build/end2endzone/win32Arduino/src/win32Arduino/arduino.cpp:926:3: error: ‘::srand’ has not been declared
     [exec]    ::srand(value);
```
I do not have enough experience for fixing the issue. Does it mean that `srand()` definition is located elsewhere? Can you take a look at it?

sidey79, 2018-04-15:
@end2endzone 
You are fast.. 

I haven't tested it, but the function you try to call, seems not to exist:

https://github.com/end2endzone/win32Arduino/blob/fce016a6cae8fc7253f549a3571e456330562e0c/src/win32Arduino/arduino.cpp#L926

Not sure, what the compiler does with this style of coding, but srand and rand should be located in 
stdlib:
http://www.cplusplus.com/reference/cstdlib/srand/
http://www.cplusplus.com/reference/cstdlib/rand/

sidey79, 2018-04-16:
Fixes are here:
#18

end2endzone, 2018-04-17:
Fixed in 623f317

closed 2018-04-17 by end2endzone