end2endzone/win32Arduino #17: Find a new name for library

end2endzone, 2018-04-15:
Once the library will be ported to linux (see issue #15 ), the name `win32Arduino` should be changed to something that does not sounds like `win32` only.

Multiple names are suggested:
* x86Arduino
* testArduino
* x86TestArduino
* UnitTestArduino

end2endzone, 2018-04-15:
The new name should not imply that the library is an Arduino emulator for PC.

sidey79, 2018-04-15:
Something like x86TestArduino sounds very good to me.

end2endzone, 2018-06-01:
@per1234 Hi. I would really like to have your feedback on this issue. I plan to rename this repository. The better suggested name (so far) is `x86TestArduino` but I still have doubt.

per1234, 2018-06-02:
I'm not very good at coming up with catchy names (as my user name will attest) but x86TestArduino seems like a reasonable choice.

It looks like a very nice project. I've been meaning to get into unit testing of Arduino code.