end2endzone/win32Arduino #21: /ci/ant/setupenv.bat is missing

sidey79, 2018-04-21:
/ci/ant/setupenv.bat is referenced in build script but not provided with repository.

end2endzone, 2018-04-24:
I am currently updating the documentation in `README.md` and `INSTALL.md`. Multiple things must be updated since the documentation outdated now that issue #15 is implemented.

To sumarize, `/ci/ant/setupenv.bat` is a generated file. The command `ant configure` generates `setupenv.bat` on Windows and `setupenv.sh` on linux. The generated file contains command to create the required environment variables.

end2endzone, 2018-04-24:
As a side note, I am still missing email notifications when someone else creates an issue. (I do receives updates when I already commented on the issue). Do you know where to activate email notifications for this features? AFAIK, everything is checked in my GitHub [settings](https://github.com/settings/notifications) page.

end2endzone, 2018-04-28:
Following changes in e14807a, the `ant configure` command now generates the file `configuration.sh` which is more straitforward.

end2endzone, 2018-04-30:
@sidey79  A new version of README.md and INSTALL.md are available. Can you confirm that documentation is clear / without ambiguity? If so, I will close the issue.

end2endzone, 2018-05-01:
@sidey79 I do apologize. Maybe it is because English isn't my first language.

The `ant configure` command is not mandatory. What is mandatory is that you must create the `WIN32ARDUINO_HOME` environment variable so that other libraries (i.e. [SoftTimers](https://github.com/end2endzone/SoftTimers) or [BitReader](https://github.com/end2endzone/BitReader)) may find this library. That is how I intent to be able to find win32Arduino from other project.

In the end, you either create the variable manually (following step 2 of Installing) or run `ant configure` followed by `configuration.sh`. But you must do one or the other.

On the raspberry pi, the content of the `configuration.sh` looks like this:
```bash
#!/bin/sh
echo Setting environment variable 'WIN32ARDUINO_HOME' to value '/home/pi/dev/github/win32Arduino'.
export WIN32ARDUINO_HOME="/home/pi/dev/github/win32Arduino"
echo Setting environment variable 'GOOGLETEST_HOME' to value '/home/pi/dev/github/win32Arduino/third_party/googletest-release-1.6.0'.
export GOOGLETEST_HOME="/home/pi/dev/github/win32Arduino/third_party/googletest-release-1.6.0"
```

It is confusing because in linux, there are commands such as `make install`? Because I did not intended to install the binary files anywhere.

end2endzone, 2018-05-01:
After looking at the documentation again, I think the command creates too much confusion. Mostly for the following reasons:
* It handles step 2 of [Installing](https://github.com/end2endzone/win32Arduino/blob/master/INSTALL.md#installing) but not other steps.
* It also handles step 2 of [Google build steps](https://github.com/end2endzone/win32Arduino/blob/master/INSTALL.md#google-c-testing-framework)
* The current documentation never states that after running `ant configure`, you must run `configuration.bat` or `configuration.sh` for actually setting the environment variables.
* The note that you refer seems to indicate that it will take care of all manual steps of Installing when it is not.

I think it would be better if I simply remove the command and leave the step 3 of Installing as it is.

end2endzone, 2018-05-01:
@sidey79 I did. See 9ebe687a27dbc5dbd078559c2a33461d33341904 and cff1c8aabbff923dc6d4d646c802c12d80c7eb53 for details.

sidey79, 2018-05-02:
I've checked your modifications.I'll try that on a vm and come back to you

end2endzone, 2018-05-27:
@sidey79 I have created tag v2.2.0. It should clear any documentation ambiguity. Please take a look at this release so I can close the issue.

In occurrence, the new release includes the following:
* No more ant scripts. Everything is done using cmake build options:  See `-DWIN32ARDUINO_BUILD_SAMPLES=ON` and `-DWIN32ARDUINO_BUILD_TEST=ON`.
* The intuitive `cmake ..` and `make` commands are now sufficient to build the library. 
* Note that `demos` are now called `samples`.

sidey79, 2018-08-08:
Sorry, I've still problems

Aas in your docs described, i've set up env variables and then did a cmake .. in the build directory on a windows system.

But, the new created Project has no dependency to the libraries:
"gtest/gtest.h": No such file or directory

end2endzone, 2018-08-10:
Hi. I am sorry for all the hassle you have been trough with the library. As the only developer on the project, I greatly lack a good testing and QA team. Also, since you switched to a windows build system, I will be more helpful to build the library.

I am assuming that you are using release [v2.2.0](https://github.com/end2endzone/win32Arduino/releases/tag/v2.2.0) or tag [v2.2.1](https://github.com/end2endzone/win32Arduino/releases/tag/v2.2.1). Could you tell me which compiler you are using? Are you using Visual Studio or something like [Cygwin or MinGW](https://stackoverflow.com/a/792142) as I only have experience with Visual Studio.

I have looked at the 'Build Steps' according to [INSTALL.md](https://raw.githubusercontent.com/end2endzone/win32Arduino/v2.2.1/INSTALL.md). There is no mention of the mandatory RAPIDASSIST_HOME environment variable. Build step 2.5 should read something like: 2.5 Define the environment variable RAPIDASSIST_HOME such that RAPIDASSIST_HOME=$WIN32ARDUINO_HOME/lib/RapidAssist. See issue #30.

If this value is not defined, the CMake should fail at configuration time. However, since you are getting the `<gtest/gtest.h>: No such file or directory` error message, I think you already figured this out.

end2endzone, 2018-08-10:
I have tried to execute the build steps on a new environement. Once you set all environment variables properly, everything is generated properly and the build should be succesful.
(Note: I am at work right now and I do not have access to git so I must download the zip file from github.com)

Here are the steps I executed:
* Downloaded tag v2.2.1 and extracted to `C:\Temp\win32Arduino-2.2.1`.
* Downloaded googletest tag [release-1.6.0](https://github.com/google/googletest/archive/release-1.6.0.zip) and extracted to `C:\Temp\win32Arduino-2.2.1\lib` and then renamed `C:\Temp\win32Arduino-2.2.1\lib\googletest-release-1.6.0` to `C:\Temp\win32Arduino-2.2.1\lib\googletest`.
* Downloaded RapidAssist tag [v0.3.4](https://github.com/end2endzone/RapidAssist/archive/v0.3.4.zip) and extracted to `C:\Temp\win32Arduino-2.2.1\lib` and then renamed `C:\Temp\win32Arduino-2.2.1\lib\RapidAssist-0.3.4` to `C:\Temp\win32Arduino-2.2.1\lib\RapidAssist`.

My directory file structure looks something like this:
```
C:\Temp\win32Arduino-2.2.1\build
C:\Temp\win32Arduino-2.2.1\build\...
C:\Temp\win32Arduino-2.2.1\...
C:\Temp\win32Arduino-2.2.1\lib\googletest
C:\Temp\win32Arduino-2.2.1\lib\RapidAssist
C:\Temp\win32Arduino-2.2.1\lib\googletest\CMakeLists.txt
C:\Temp\win32Arduino-2.2.1\lib\googletest\include
C:\Temp\win32Arduino-2.2.1\lib\googletest\src
C:\Temp\win32Arduino-2.2.1\lib\googletest\test
C:\Temp\win32Arduino-2.2.1\lib\googletest\...
C:\Temp\win32Arduino-2.2.1\lib\RapidAssist\include
C:\Temp\win32Arduino-2.2.1\lib\RapidAssist\lib
C:\Temp\win32Arduino-2.2.1\lib\RapidAssist\src
C:\Temp\win32Arduino-2.2.1\lib\RapidAssist\test
C:\Temp\win32Arduino-2.2.1\lib\RapidAssist\...
```
(note that many files and folders are ommited with `...`)

* Open a command prompt and enter the following commands:
```
set WIN32ARDUINO_HOME=C:\Temp\win32Arduino-2.2.1
set GOOGLETEST_HOME=%WIN32ARDUINO_HOME%\lib\googletest
set RAPIDASSIST_HOME=%WIN32ARDUINO_HOME%\lib\RapidAssist
cd /d C:\Temp\win32Arduino-2.2.1
mkdir build
cd build
cmake ..
```

Using only these commands, I get the following displayed on the console:
```
Microsoft Windows [Version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation.  All rights reserved.

M:\>set WIN32ARDUINO_HOME=C:\Temp\win32Arduino-2.2.1

M:\>set GOOGLETEST_HOME=%WIN32ARDUINO_HOME%\lib\googletest

M:\>set RAPIDASSIST_HOME=%WIN32ARDUINO_HOME%\lib\RapidAssist

M:\>cd /d C:\Temp\win32Arduino-2.2.1

C:\Temp\win32Arduino-2.2.1>mkdir build

C:\Temp\win32Arduino-2.2.1>cd build

C:\Temp\win32Arduino-2.2.1\build>cmake ..
-- Building for: Visual Studio 10 2010
-- The C compiler identification is MSVC 16.0.40219.1
-- The CXX compiler identification is MSVC 16.0.40219.1
-- Check for working C compiler using: Visual Studio 10 2010
-- Check for working C compiler using: Visual Studio 10 2010 -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler using: Visual Studio 10 2010
-- Check for working CXX compiler using: Visual Studio 10 2010 -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_BINARY_DIR:         C:/Temp/win32Arduino-2.2.1/build
-- CMAKE_CURRENT_BINARY_DIR: C:/Temp/win32Arduino-2.2.1/build
-- CMAKE_SOURCE_DIR:         C:/Temp/win32Arduino-2.2.1
-- CMAKE_CURRENT_SOURCE_DIR: C:/Temp/win32Arduino-2.2.1
-- PROJECT_BINARY_DIR:       C:/Temp/win32Arduino-2.2.1/build
-- PROJECT_SOURCE_DIR:       C:/Temp/win32Arduino-2.2.1
-- EXECUTABLE_OUTPUT_PATH:   C:/Temp/win32Arduino-2.2.1/build/bin
-- LIBRARY_OUTPUT_PATH:      C:/Temp/win32Arduino-2.2.1/build/bin
-- CMAKE_MODULE_PATH:
-- CMAKE_COMMAND:            D:/dev/OpenSceneGraph/cmake/cmake-3.4.3-win32-x86/bin/cmake.exe
-- CMAKE_ROOT:               D:/dev/OpenSceneGraph/cmake/cmake-3.4.3-win32-x86/share/cmake-3.4
-- CMAKE_CURRENT_LIST_FILE:  C:/Temp/win32Arduino-2.2.1/CMakeLists.txt
-- CMAKE_CURRENT_LIST_LINE:  20
-- CMAKE_INCLUDE_PATH:
-- CMAKE_LIBRARY_PATH:
-- CMAKE_SYSTEM:             Windows-6.1.7601
-- CMAKE_SYSTEM_NAME:        Windows
-- CMAKE_SYSTEM_VERSION:     6.1.7601
-- CMAKE_SYSTEM_PROCESSOR:   AMD64

WIN32ARDUINO_VERSION_STRING: 2.3.0

-- PROJECT_NAME:             RapidAssist
-- CMAKE_BINARY_DIR:         C:/Temp/win32Arduino-2.2.1/build
-- CMAKE_CURRENT_BINARY_DIR: C:/Temp/win32Arduino-2.2.1/build/lib/RapidAssist
-- CMAKE_SOURCE_DIR:         C:/Temp/win32Arduino-2.2.1
-- CMAKE_CURRENT_SOURCE_DIR: C:/Temp/win32Arduino-2.2.1/lib/RapidAssist
-- PROJECT_BINARY_DIR:       C:/Temp/win32Arduino-2.2.1/build/lib/RapidAssist
-- PROJECT_SOURCE_DIR:       C:/Temp/win32Arduino-2.2.1/lib/RapidAssist
-- EXECUTABLE_OUTPUT_PATH:   C:/Temp/win32Arduino-2.2.1/build/bin
-- LIBRARY_OUTPUT_PATH:      C:/Temp/win32Arduino-2.2.1/build/bin
-- CMAKE_MODULE_PATH:
-- CMAKE_COMMAND:            D:/dev/OpenSceneGraph/cmake/cmake-3.4.3-win32-x86/bin/cmake.exe
-- CMAKE_ROOT:               D:/dev/OpenSceneGraph/cmake/cmake-3.4.3-win32-x86/share/cmake-3.4
-- CMAKE_CURRENT_LIST_FILE:  C:/Temp/win32Arduino-2.2.1/lib/RapidAssist/CMakeLists.txt
-- CMAKE_CURRENT_LIST_LINE:  21
-- CMAKE_INCLUDE_PATH:
-- CMAKE_LIBRARY_PATH:
-- CMAKE_SYSTEM:             Windows-6.1.7601
-- CMAKE_SYSTEM_NAME:        Windows
-- CMAKE_SYSTEM_VERSION:     6.1.7601
-- CMAKE_SYSTEM_PROCESSOR:   AMD64

RAPIDASSIST_VERSION_STRING: 0.3.4

-- Configuring done
-- Generating done
-- Build files have been written to: C:/Temp/win32Arduino-2.2.1/build

C:\Temp\win32Arduino-2.2.1\build>
```

note that win32Arduino version is incorrectly reported as 2.3.0. This is a known issue (#31).

I also expect my work computer to be a new 'environment' which should not have pre-existing mandatory configuration artifact from my other projects.

end2endzone, 2018-08-10:
Running `cmake --build . --config Release`, I get the following displayed on the console:
```
C:\Temp\win32Arduino-2.2.1\build>cmake --build . --config Release
Microsoft (R) Build Engine version 4.0.30319.17929
[Microsoft .NET Framework, version 4.0.30319.18063]
Copyright (C) Microsoft Corporation. All rights reserved.

Build started 8/10/2018 7:40:41 AM.
Project "C:\Temp\win32Arduino-2.2.1\build\ALL_BUILD.vcxproj" on node 1 (default targets).
Project "C:\Temp\win32Arduino-2.2.1\build\ALL_BUILD.vcxproj" (1) is building "C:\Temp\win32Arduino-2.2.1\build\ZERO_CHECK.vcxproj" (2) on node 1 (default targets).
PrepareForBuild:
  Creating directory "Win32\Release\ZERO_CHECK\".
  Creating directory "Release\".
InitializeBuildStatus:
  Creating "Win32\Release\ZERO_CHECK\ZERO_CHECK.unsuccessfulbuild" because "AlwaysCreate" was specified.
CustomBuild:
  Checking Build System
  CMake does not need to re-run because C:/Temp/win32Arduino-2.2.1/build/CMakeFiles/generate.stamp is up-to-date.
  CMake does not need to re-run because C:/Temp/win32Arduino-2.2.1/build/src/win32Arduino/CMakeFiles/generate.stamp is up-to-date.
  CMake does not need to re-run because C:/Temp/win32Arduino-2.2.1/build/lib/RapidAssist/CMakeFiles/generate.stamp is up-to-date.
  CMake does not need to re-run because C:/Temp/win32Arduino-2.2.1/build/lib/RapidAssist/src/rapidassist/CMakeFiles/generate.stamp is up-to-date.
FinalizeBuildStatus:
  Deleting file "Win32\Release\ZERO_CHECK\ZERO_CHECK.unsuccessfulbuild".
  Touching "Win32\Release\ZERO_CHECK\ZERO_CHECK.lastbuildstate".
Done Building Project "C:\Temp\win32Arduino-2.2.1\build\ZERO_CHECK.vcxproj" (default targets).

Project "C:\Temp\win32Arduino-2.2.1\build\ALL_BUILD.vcxproj" (1) is building "C:\Temp\win32Arduino-2.2.1\build\lib\RapidAssist\src\rapidassist\rapidassist.vcxproj" (3) on node 1 (default targets).
PrepareForBuild:
  Creating directory "rapidassist.dir\Release\".
  Creating directory "C:\Temp\win32Arduino-2.2.1\build\bin\Release\".
InitializeBuildStatus:
  Creating "rapidassist.dir\Release\rapidassist.unsuccessfulbuild" because "AlwaysCreate" was specified.
CustomBuild:
  Building Custom Rule C:/Temp/win32Arduino-2.2.1/lib/RapidAssist/src/rapidassist/CMakeLists.txt
  CMake does not need to re-run because C:\Temp\win32Arduino-2.2.1\build\lib\RapidAssist\src\rapidassist\CMakeFiles\generate.stamp is up-to-date.
MakeDirsForCl:
  Creating directory "C:\Temp\win32Arduino-2.2.1\build\lib\RapidAssist\src\rapidassist\Release".
ClCompile:
  C:\Program Files (x86)\Microsoft Visual Studio 10.0\VC\bin\CL.exe /c /I"C:\Temp\win32Arduino-2.2.1\lib\RapidAssist\include" /I"C:\Temp\win32Arduino-2.2.1\lib\googletest\include" /nologo /W3 /WX- /O2 /Ob2 /Oy- /D WIN32 /D _WINDOWS /D NDEBUG /D _CRT_SECURE_NO_WARNINGS /D "CMAKE_INTDIR=\"Release\"" /D _MBCS /Gm- /EHsc /MD /GS /fp:precise /Zc:wchar_t /Zc:forScope /GR /Fo"rapidassist.dir\Release\\"
  /Fd"rapidassist.dir\Release\vc100.pdb" /Gd /TP /analyze- /errorReport:queue ..\..\..\..\..\lib\RapidAssist\src\rapidassist\cli.cpp ..\..\..\..\..\lib\RapidAssist\src\rapidassist\cppencoder.cpp ..\..\..\..\..\lib\RapidAssist\src\rapidassist\environment.cpp ..\..\..\..\..\lib\RapidAssist\src\rapidassist\filesystem.cpp ..\..\..\..\..\lib\RapidAssist\src\rapidassist\gtesthelp.cpp ..\..\..\..\..\lib
  \RapidAssist\src\rapidassist\logger.cpp ..\..\..\..\..\lib\RapidAssist\src\rapidassist\random.cpp ..\..\..\..\..\lib\RapidAssist\src\rapidassist\time_.cpp ..\..\..\..\..\lib\RapidAssist\src\rapidassist\strings.cpp
  cli.cpp
  cppencoder.cpp
  environment.cpp
  filesystem.cpp
  gtesthelp.cpp
  logger.cpp
  random.cpp
  time_.cpp
  strings.cpp
  Generating Code...
Lib:
  C:\Program Files (x86)\Microsoft Visual Studio 10.0\VC\bin\Lib.exe /OUT:"C:\Temp\win32Arduino-2.2.1\build\bin\Release\rapidassist.lib" /NOLOGO rapidassist.dir\Release\cli.obj
  rapidassist.dir\Release\cppencoder.obj
  rapidassist.dir\Release\environment.obj
  rapidassist.dir\Release\filesystem.obj
  rapidassist.dir\Release\gtesthelp.obj
  rapidassist.dir\Release\logger.obj
  rapidassist.dir\Release\random.obj
  rapidassist.dir\Release\time_.obj
  rapidassist.dir\Release\strings.obj  /machine:X86
  rapidassist.vcxproj -> C:\Temp\win32Arduino-2.2.1\build\bin\Release\rapidassist.lib
FinalizeBuildStatus:
  Deleting file "rapidassist.dir\Release\rapidassist.unsuccessfulbuild".
  Touching "rapidassist.dir\Release\rapidassist.lastbuildstate".
Done Building Project "C:\Temp\win32Arduino-2.2.1\build\lib\RapidAssist\src\rapidassist\rapidassist.vcxproj" (default targets).

[...]

FinalizeBuildStatus:
  Deleting file "Win32\Release\ALL_BUILD\ALL_BUILD.unsuccessfulbuild".
  Touching "Win32\Release\ALL_BUILD\ALL_BUILD.lastbuildstate".
Done Building Project "C:\Temp\win32Arduino-2.2.1\build\ALL_BUILD.vcxproj" (default targets).

Build succeeded.
    0 Warning(s)
    0 Error(s)

Time Elapsed 00:00:07.03

C:\Temp\win32Arduino-2.2.1\build>
```

end2endzone, 2018-08-10:
If you are still having issues, then it might be one of the following:
1) On Windows, declaring environment variable from the [System Properties, Environment Variable](https://www.nextofwindows.com/how-to-addedit-environment-variables-in-windows-7) does not propagate to existing processes (including opened command prompt). You need to log out and back in to force all new processes to use the value. This is also why I forced the values within the command prompt using the `set` command.
2) Missing or incorrect submodule. I am using [git submodules](https://git-scm.com/docs/git-submodule) for automatically downloading a compatible version of `googletest` and `RapidAssist`. Did you try running `git submodule update --init --recursive` at the root of the repository?
3) Obviously, the file gtest.h is missing (see point 2) or you have the wrong path to reach gtest.h. It should be located at `C:\Temp\win32Arduino-2.2.1\lib\googletest\include\gtest\gtest.h`. Assuming you are using Visual Studio, you can take a look at the generated projects files that have a dependency to googletest. That is `C:\Temp\win32Arduino-2.2.1\build\lib\RapidAssist\src\rapidassist\rapidassist.vcxproj` or `C:\Temp\win32Arduino-2.2.1\build\test\win32Arduino_unittest.vcxproj`. Those are XML files. Look for tags `<AdditionalIncludeDirectories>`. You should see something like this:
```
      <AdditionalIncludeDirectories>C:\Temp\win32Arduino-2.2.1\include;C:\Temp\win32Arduino-2.2.1\lib\RapidAssist\include;C:\Temp\win32Arduino-2.2.1\lib\googletest\include;%(AdditionalIncludeDirectories)</AdditionalIncludeDirectories>
```

which specify where to find the googletest include folder.
4) Try resolving your issue looking at the [appveyor](https://github.com/end2endzone/win32Arduino/blob/v2.2.1/appveyor.yml)'s build configuration file. It defines all steps required to build the library on Windows from a 'plain' environment. (you will have to go through `/ci/appveyor/` build scripts).

Let me know if it resolved your issue.
Antoine.

end2endzone, 2018-08-12:
FYI, I have created [v2.2.2](https://github.com/end2endzone/win32Arduino/releases/tag/v2.2.2) release.

sidey79, 2018-08-24:
okay, i've retested with v2.2.2  and i've got it now to build your project :)

But i think, in the docs for 2.2.2 it is missing that three environment variables are needed and that downloading rapidAssist is a requirement.

I'tt next try to build my tests from my project

end2endzone, 2018-08-27:
> okay, i've retested with v2.2.2 and i've got it now to build your project

That is good news!

> it is missing that three environment variables are needed 

I don't understand your comment about environment variables. The 3 required environment variables are actually specified in [INSTALL.md, line 62](https://github.com/end2endzone/win32Arduino/blame/v2.2.2/INSTALL.md#L62) and above.

Do I miss something ?

> downloading rapidAssist is a requirement

Only if you downloaded the source code from github's zip file. If you cloned the repository, the expected version of RapidAssist should have been cloned as well. See [this comment](https://github.com/end2endzone/win32Arduino/issues/32#issuecomment-416295018) for explanation and [this comment](https://github.com/end2endzone/win32Arduino/issues/32#issuecomment-416300875) in issue #32 for a proposed solution.

sidey79, 2018-08-27:
May i read the install.md from master or prerelease2.2  in the tagged version it it clear described.

The thing with the submodule wasn't clear to me. I've cloned your repository alway and now i checked out the tag 2.2.2, but did not knew about the submodules.

I had gtest already deployed in another directory so i deployed rapidassist also to another directory.
I've checked the thing with the submodule, when i update the submodules, then the dependent libs are deployed into the libs/ directory.

So may a hint, that the submodules are located there could solve that issue.

end2endzone, 2018-08-29:
I pushed a new commit (bfd8c1198ac52861ef09811b8d6589cc60018f97). This is a documentation update that should clarify new build instructions for Windows and Linux and remove ambiguity about submodules and dependencies. If should fix issue #21, issue #32 and issue #33. Please tell me if this change has resolved your issue and I will close it.

end2endzone, 2018-08-29:
I will close this issue since the submodule issue is now clarified.

If you still have concerns/questions, please open a new issue. I don't want this issue to span/cover more than one subject (the title does not reflect your current concerns anymore)

closed 2018-08-29 by end2endzone