end2endzone/win32Arduino #32: rapidassist outside of win32arduino?

sidey79, 2018-08-24:
RapidAssist can not be deployed into another path. But it is also not a submobule of win32Arduin repository and i haven't found the depency on that in install.md.

If i place rapidassist in another directory the following occures:

after cmake..  the following error occures:

WIN32ARDUINO_VERSION_STRING: 2.2.2

CMake Error at CMakeLists.txt:109 (add_subdirectory):
  add_subdirectory not given a binary directory but the given source
  directory
  "...../Project/tests/RapidAssist" is not
  a subdirectory of
  "...../Project/tests/win32arduino".  When
  specifying an out-of-tree source a binary directory must be explicitly
  specified.

So one option is, to implement rapidassist as a submodule in this repository

end2endzone, 2018-08-27:
Regarding your specific error message, you seem to be having the same issue as [this guy](https://stackoverflow.com/questions/50408169/cmake-error-add-subdirectory-not-given-a-binary-directory). You could try the suggestion and hope you can compile the code.

end2endzone, 2018-08-27:
RapidAssit is already a submodule of win32Arduino. See [the github code page](https://github.com/end2endzone/win32Arduino/tree/v2.2.2/lib) for reference. You can see that googletest and RapidAssist are submodule and pointing to specific commits of their respective repository.

However, when you [download the source code](https://github.com/end2endzone/win32Arduino/archive/v2.2.2.zip) from github, the downloaded zip file does not contains the submodules but empty directories. In this case, you must manually download the *specific* version of RapidAssit and yes **RapidAssist should be considered as dependency**.

But if you used `git clone` to "download" the source code, then the appropriate version of RapidAssist should be automatically downloaded with `git submodule update`. In this case, the user should not have to deal with downloading a particular version of RapidAssist. Everything is done automatically. In this case, **RapidAssist should *not* be considered as a dependency** (IMHO).

I am having a hard time dealing with git submodules and (obviously) properly documenting the dependency section of install.md. Coming from subversion world, I am used to create "pointers to other commit of a repository" (called externals) in order to guide the user to know that a repository is "compatible" with another repository.

This should be clarified in the install.md documentation.

sidey79, 2018-08-27:
With this proposed documentation it is much clearer i think.

end2endzone, 2018-08-29:
I pushed a new commit (bfd8c1198ac52861ef09811b8d6589cc60018f97). This is a documentation update that should clarify new build instructions for Windows and Linux and remove ambiguity about submodules and dependencies. If should fix issue #21, issue #32 and issue #33. Please tell me if this change has resolved your issue and I will close it.

sidey79, 2018-08-29:
Wenn can close this issue. Thanks for clarify the dependencys.