end2endzone/win32Arduino #35: cmake doesn't find GTEST via find_package

sidey79, 2018-08-30:
On master i've tryed building win32arduino, but it doesn't work, because it is not finding gtest.

I've searched a little bit, how to register gtest to cmake, but i have no idea how to do that:

```
CMake Error at C:/Program Files/CMake/share/cmake-3.12/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR
  GTEST_MAIN_LIBRARY)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.12/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.12/Modules/FindGTest.cmake:196 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:125 (find_package)
```

end2endzone, 2018-08-31:
Do you know if gtest was properly installed on your system? The two possible issues comes to mind:
1) You did not run the `INSTALL` project when you compiled gtest. If you actually ran the INSTALL target, you should have a new directory under `C:/Program Files (x86)/googletest-distribution` or maybe `C:/Program Files (x86)/gtest`, I can't exactly remember.
2) If you specified a custom install location for gtest with `-DCMAKE_INSTALL_PREFIX=<somedirectory>` at cmake at configure time (I usually do), look to see if the actual directory was created. Again, execute the `INSTALL` target to create the directory.

The install directory should contains the `include` and a `lib` directories.

If gtest was properly installed, then you need to tell CMake where to find gtest installation directory.
1) If you used option 1, then you should not have to specify where gtest is installed. CMake should be able to find gtest without difficulties.
2) If you used a custom installation directory for gtest, you must tell CMake where to find the installation directory. This can be done with GTEST_ROOT environment variable. This is a special variable that `find_package()` will use to find the installation directory. In other words, before running `cmake ..` for win32Arduino,  run the command `set GTEST_ROOT=<somedirectory>` and try `cmake ..` again. This should allow CMake to find gtest dependency.

end2endzone, 2018-08-31:
If my previous suggestion fails, take a look at [appveyor's scripts](https://github.com/end2endzone/win32Arduino/tree/master/ci/appveyor). Look particularly at `install_googletest.bat`, `install_rapidassist.bat` and `build_library.bat`. These are the scripts that must actually executed on a new window system. You will see that I use `-DCMAKE_INSTALL_PREFIX=<somedirectory>` extensively.

end2endzone, 2018-08-31:
As a general rule, if `find_package(foobar)` fails, and you know where `foobar` was installed, you can usually define the environment variable `foobar_DIR` and make it point to the installation directory. This mechanism is already built-it CMake.

Note: I see you are using `cmake-3.12`. I am actually using `CMake 3.4.3`. Newer version of CMake seem to be requesting `foobar_ROOT` environment variable [according to the latest win32Arduino build](https://ci.appveyor.com/project/end2endzone/win32arduino#L453). This is new since 3.4.3. However, CMake is to be able to find `GTest` and `rapidassist` packages with the `foobar_DIR` format.

sidey79, 2018-08-31:
sorry, i am an noob regarding to cmake. I Looked to your `install_googletest.bat` that helped me a little bit to understand what to do.

I'm now here:

```
cmake -DCMAKE_INSTALL_PREFIX=%GTEST_ROOT% -Dgtest_force_shared_crt=ON  ..
cmake --build . --config Release

```
Till that point, i got anything working. But now when i try to install

```
cmake --build . --config Release --target install
Microsoft (R)-Build-Engine, Version 15.8.166+gd4e8d81a88 für .NET Framework
Copyright (C) Microsoft Corporation. Alle Rechte vorbehalten.

MSBUILD : error MSB1009: Die Projektdatei ist nicht vorhanden.
Schalter: install.vcxproj
```

It doesn't find any install.vcproj  which isn't a suprise to me, because this file doesn't exists.
Also the GTEST_ROOT directory is empty

This seems to be an GTest issue i know, but otherwise i didn't find any "install" command on the googletest site.

end2endzone, 2018-09-01:
If I understand you correctly, when you tried
```cmake
cmake -DCMAKE_INSTALL_PREFIX=%GTEST_ROOT% -Dgtest_force_shared_crt=ON  ..
cmake --build . --config Release
```
you were actually trying to build gtest and not win32Arduino.

If not, then what are actually saying with `cmake -DCMAKE_INSTALL_PREFIX=%GTEST_ROOT% -Dgtest_force_shared_crt=ON  ..` is that you want to install win32Arduino into gtest install directory. This **could** be working but is considered bad practice.

sidey79, 2018-09-01:
Yes, i tried to build and Install gtest. Isn't that correct?

end2endzone, 2018-09-01:
Yes that is correct.

end2endzone, 2018-09-01:
Regarding your problem with the missing `install.vcxproj`, I do not know what to think. As far as I know, [CMake always creates an INSTALL target](https://stackoverflow.com/questions/26677454/cmake-does-not-create-install-project-in-visual-studio) when your CMakeList.txt file contains `install()`. The googletest's CMakeList.txt [does contains install() commands](https://github.com/google/googletest/blob/ec44c6c1675c25b9827aacd08c02433cccde7780/googletest/CMakeLists.txt#L105). 

Are you sure that `%GTEST_ROOT%` is properly set for your current process? To verify, try to execute `echo GTEST_ROOT=%GTEST_ROOT%` and look at the result. If `%GTEST_ROOT%` is unset, then it would mean you are actually running `cmake -DCMAKE_INSTALL_PREFIX= -Dgtest_force_shared_crt=ON  ..` (note the empty CMAKE_INSTALL_PREFIX), which could prevent CMake from generating the INSTALL target.

One more thing, the target should be INSTALL and not install. As far as I know, Windows/Visual Studio should not be case sensitive but you never know...

Another option is difference between Visual Studio versions. I **think** you are using Visual Studio 2017 (based on `Microsoft (R)-Build-Engine, Version 15`) and I also think that VS2017 has its own version of CMake. Could there be a difference ? I build locally with VS2010 and AppVeyor builds with VS2015 (see [this line](https://ci.appveyor.com/project/end2endzone/win32arduino#L33) and [this line](https://ci.appveyor.com/project/end2endzone/win32arduino#L203)).

sidey79, 2018-09-01:
ok, i got it.

there are two directorys called googletest.

I was trying all that stuff in the subdir googletest of googletest....
`win32arduino\lib\googletest\googletest`

I then changed to the upper directory and installed googletest
```
cd build
cmake ..
cmake -DBUILD_GMOCK=0 -DCMAKE_INSTALL_PREFIX=%GTEST_ROOT% -Dgtest_disable_pthreads=on -Dgtest_force_shared_crt=ON  ..
cmake --build . --config Release
cmake --build . --config Release --target install
```

Googletest is now installed.

closed 2018-09-01 by sidey79

end2endzone, 2018-09-04:
> there are two directorys called googletest.
> I was trying all that stuff in the subdir googletest of googletest....
> win32arduino\lib\googletest\googletest
> I then changed to the upper directory and installed googletest

Correct!

But I see you execute cmake configuration twice:
```
cmake ..
cmake -DBUILD_GMOCK=0 -DCMAKE_INSTALL_PREFIX=%GTEST_ROOT% -Dgtest_disable_pthreads=on -Dgtest_force_shared_crt=ON  ..
```

I am not even sure your second cmake call does something. If I try this I get the following warnings from CMake:
```
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:
    gtest_disable_pthreads
    gtest_force_shared_crt
```
and my solution file is missing both gtest and gmock.

end2endzone, 2018-09-04:
I noticed that you call `-DBUILD_GMOCK=0`. Note that CMake boolean option should be specified to `ON` of `OFF`. I do not think that `0` (or `1`) are accepted values. I think that when you call `-DBUILD_GMOCK=0`, this results in an empty solution. See [How to set a CMake option() at command line](https://stackoverflow.com/questions/18435516/how-to-set-a-cmake-option-at-command-line) and [How the heck does one set options](https://cmake.org/pipermail/cmake/2011-November/047168.html) for reference.

sidey79, 2018-09-04:
When specifyin the CMAKE_INSTALL_PREFIX to a directory Documents\project\tests\install

```
cmake -DBUILD_GMOCK=0 -DCMAKE_INSTALL_PREFIX=%GTEST_ROOT%  -Dgtest_force_shared_crt=ON  ..
cmake --build . --config Release --target install
```

then cmake must know this path, to look there for installed projects which doen't contain a find file.
`set CMAKE_PREFIX_PATH=Documents\project\tests\install
`

sidey79, 2018-09-09:
Just a info for you.
I got it now, that my testproject and all the libs compiles on linux (travis-ci) and also on my windows machine.

The vs project , which cmake generates was easy to add to my vs project. On linux i hat do debug some things but now it is up and running